ecmnet/MAVGCL #8: Import PX4Log files

ecmnet, 2016-02-12:


mhkabir, 2016-03-17:
@ecmnet Any chance that you will be able to implement this soon? :) It would be great to have the ability to replay px4log files :)

Thanks!

ecmnet, 2016-03-17:
@mhkabir This is quite a difficult question. Currently MAVGCL uses a flat data model structure (with hardcoded key figures), which I took over from my previous flight stack implementation. Now, I am not sure whether to keep that model as a dynamic one would give more flexibility. Since this question is tightly coupled to the mavlink-parser, I use, I thought about waiting for MAVLINK 2.0, before changing that. On the other hand, it would be quite easy to keep the current model and assign the key-figures of the px4log to that flat model. So, to be honest, I just don't know what to do here. What is your opinion?

mhkabir, 2016-03-17:
@ecmnet Mavlink 2.0 is still sometime away. I think it would be prudent to keep using the current model (hey, it works well!) till we can get a stable MAVLink 2.0 spec. 
There is practically nothing we can use to replay logs currently, and MAVGCL is the best option for this to date. If we can get log-replay working, it would also increase the user base for this application.

ecmnet, 2016-03-17:
@mhkabir Basic PX4Log Import implemented. Currently I have mapped only L.NEDX,Y,Z. Further mapping to do in https://github.com/ecmnet/MAVComm/blob/master/MAVComm/src/com/comino/msp/model/MSTYPE.java

mhkabir, 2016-03-17:
@ecmnet Awesome :) How can I test? What's the best way to help you add more support?

ecmnet, 2016-03-17:
@mhkabir just run the app in any mode, open px4file and select LocalPosition. The screenshot shows an uploaded px4log:

[image]

ecmnet, 2016-03-17:
Note: As the sampling in MAVGCL is currently fixed to 50ms, px4logs will be read accordingly and thus provide NOT the full resolution in time

ecmnet, 2016-03-17:
@mhkabir Is there any easy readable list of the px4log abbreviations available ? This would make mapping more easy for me.

mhkabir, 2016-03-17:
@ecmnet Does this help? https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2_messages.h

mhkabir, 2016-03-17:
More specifically, https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2_messages.h#L614-L673

ecmnet, 2016-03-17:
@mhkabir Yes. Thanks a lot!

mhkabir, 2016-03-18:
Thanks for this. I pulled master, and built it, but I can't find a way to
import a file...

This must be something silly on my part, but what do I press to be able to
load the log?

ecmnet, 2016-03-18:
@mhkabir Hm. Should be under the menu File -> Import px4Log

ecmnet, 2016-03-18:
It's a bit weird. JavaFX does not use the OS X Menu-Line:

[image]

I just looked around and there seems to be a fix for this.

ecmnet, 2016-03-18:
@mhkabir Btw. Let me know, which additional key-figures (either from MAVLink or from px4Log) would be useful for you...

ecmnet, 2016-03-20:
@mhkabir SystemMenu issue fixed.

closed 2016-03-20 by ecmnet