ecmnet/MAVGCL #11: The class path should not be hardcoded

LorenzMeier, 2016-02-13:
You're currently hardcoding the class path:
```xml
<taskdef resource="com/sun/javafx/tools/ant/antlib.xml" uri="javafx:com.sun.javafx.tools.ant" classpath="/Library/Java/JavaVirtualMachines/jdk1.8.0_51.jdk/Contents/Home/lib/ant-javafx.jar" />
```

I'm sure there is a proper way in Java to do this without hardcoding a specific location and Java version. This path will not work on Linux and will not work on any other OS X machine (it does not work on my Mac).

LorenzMeier, 2016-02-13:
If you can resolve this I can be of more help testing this. I spent 15 mins trying to work out how to fix it properly.

ecmnet, 2016-02-14:
Fix provided, that at least works on OSX. To be checked for other platforms. Closing meanwhile.

closed 2016-02-14 by ecmnet