ecmnet/MAVGCL #14: UDP connection issues

ecmnet, 2016-02-14:
Think of rewriting the UDP communication driver to address the following issues:

- [ ] Bind to all local addresses, including broadcast and loopback
- [ ] Once the first heartbeat packet is received, send back a heartbeat packet
- [ ] Move from non-blocking mode to blocking mode to reduce CPU burning
- [ ] Fix reconnection issue, when interface is removed and re-inserted from/to PC (e.g. USB-WLAN adapter)
- [ ] Reduce latency when reconnecting
- [ ] Allow parallel connection with QGC

LorenzMeier, 2016-02-14:
Can't connect to SITL if setting listen address to 127.0.0.1. The network API really needs to be fixed before this is ready for wider use.

ecmnet, 2016-02-14:
I agree. SITL uses 14556 instead of 14555.

ecmnet, 2016-03-06:
Refer to https://github.com/ecmnet/MAVComm/issues/1

LorenzMeier, 2016-06-23:
It would be great if the app would ask on boot which IP to connect to and remember the last IP it was connected to. Then it will be really easy to have that as default. The current approach with the IP in the build file doesn't really scale.

ecmnet, 2016-06-24:
I am currently working on some 'Preferences'. This could also cover the IP issue.

ecmnet, 2016-06-24:
@LorenzMeier I added a preferences dialog (File -> Preferences) to setup IP address and port (Restart required). Closing this for now.

closed 2016-06-24 by ecmnet