ecmnet/MAVGCL #17: Connection established, but no data graphed

LorenzMeier, 2016-02-15:
I can see the data in the inspector, but I can't graph anything. Might well be the height issue with the UI.

[image]

ecmnet, 2016-02-15:
Did you press 'recording' or enabled an automatic trigger?

LorenzMeier, 2016-02-15:
No, and it wasn't obvious I have to. I would advise to start it automatically and allow to configure it in the panel. Its like if your oscilloscope would show nothing on boot - that would be very confusing.

ecmnet, 2016-02-15:
That's why the button is in red 😄. But I agree, will change that.

LorenzMeier, 2016-02-15:
The button is not red. If it would have been red I would probably have seen it. Its also not obvious its related to the graph. Change "recording" to "capture". Recording means to store something to a disk, which is not happening here.

ecmnet, 2016-02-15:
In your screenshot, the button is red. Concerning texts, feel free to create a pull and change it. For me this has quite low priority as it is a question of interpretation.

LorenzMeier, 2016-02-15:
You're expectation bias makes the button red for you. But in fact the button is grey with a dark red font. Because you designed it you believe its very visible, but its not.

ecmnet, 2016-02-15:
You are right. This is a grey button, accentuated with a red font. It was the accentuation, I was aiming at with a smile some lines above. A bit of a sophistry here.

closed 2016-02-15 by ecmnet