ecmnet/MAVGCL #43: Request: BIN version

tubeme, 2016-04-20:
Hi,

I would love to try your program but I have no compile environment. Is there a way you could publish Windows Bin version?

Thanks

ecmnet, 2016-04-21:
Sorry, but this currently is not possible as during packaging your environment needs to be included (e.g. IP address, etc).

ecmnet, 2016-04-22:
I will provide binaries as soon as settings are customizable in preferences.

ecmnet, 2016-04-22:
Just for you to try it. You might have to adjust the peerAddress in the .bat file...

[MAVGCL Windows.zip](https://github.com/ecmnet/MAVGCL/files/232199/MAVGCL.Windows.zip)

tubeme, 2016-05-03:
Thanks a lot!! There is some problem with the java version I suppose. Here is the error:

E:\DRONES\Analisys Tool>java -jar MAVGAnalysis.jar --peerAddress=172.168.178.1
Exception in thread "main" java.lang.UnsupportedClassVersionError: com/comino/flight/MainApp : Unsupported major.minor version 52.0
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(Unknown Source)
        at java.security.SecureClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.access$100(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.launcher.LauncherHelper.checkAndLoadMain(Unknown Source)

I have JRE 7 update 80 64 bit.

Any Ideas?

ecmnet, 2016-05-04:
Due to lambdas, Java 8 is required.

closed 2016-06-18 by ecmnet

ecmnet, 2016-06-26:
@tubeme Binaries available now. Include required JRE.