ecmnet/MAVGCL #47: Request: presets for tuning

dagar, 2016-06-17:
This is a mockup I put together for what I eventually wanted to implement in QGroundControl for PID tuning. Would you be interested in doing something like this for MAVGCL? I could certainly help.

[image]

https://github.com/mavlink/qgroundcontrol/issues/2878

ecmnet, 2016-06-17:
@dagar Should not be too difficult, as MAVGCL has its own data model storing key figures. Should the graph base on MAVlink data for in-flight display? (I am not aware of topics which represent 'command' in your mock-up). Response, I assume, are actual angles, etc. Are those key figures already in the PX4Log? In that case it would be easy to add them to the model, if you provide me the (sometimes cryptic) key-figure names.

dagar, 2016-06-17:
Initially I was thinking just mavlink for tuning in flight, but I'd be very impressed if it could be abstracted to work with mavlink and a px4log.

Mavlink is a bit inconsistent with the naming. I just called it command here, but I don't personally care if it's called command/setpoint/target.

There's a little bit of conversion to be human friendly and intuitive. I think everything should probably be displayed in degrees. Most of these are in radians, and the attitude target is a quaternion.

For the displayed mockup it would use -
ATTITUDE - http://mavlink.org/messages/common#ATTITUDE
ATTITUDE_TARGET - http://mavlink.org/messages/common#ATTITUDE_TARGET

Actuator would be from ACTUATOR_CONTROL_TARGET.

The TECS tab (altitude, throttle controller tuning) is weirder because the mavlink spec only has altitude error and airspeed error. We could combine that with airspeed and altitude to plot command/response in the same way.

Then there's showing the appropriate params for each tuning tab. Maybe we could just stupidly filter as these things can change over time.
Pitch: FW_P*, etc

ecmnet, 2016-06-17:
@dagar I'll have a look on this tomorrow. Most of it needs to be done in MAVComm, as the data model and the MAVLink parser is implemented there. PX4 log, of course, does not work in-flight and is very slow, so I think it makes sense to stick to MAVLink in the first step. 

Regarding the params, I am not quite sure, whether an implementation as you suggest makes sense, as in MAVGVL params are shown as tree in a separate tab, so switching to that tab and scrolling to the corresponding tree items should not be too inconvenient. Maybe a filter in this view is sufficient. For in-flight tuning, The best way would be to adjust PIDs via RC.

ecmnet, 2016-06-18:
@dagar by now I added the following key figures (transforming them into degrees):
   Roll, Pitch, Yaw, RollRate, PitchRate, YawRate, SetPoint Roll, SetPoint Pitch, SetPoint Yaw

dagar, 2016-06-18:
You're fast. Two simultaneous plots is probably sufficient. Seeing what the actuators are doing isn't critical.

I agree the best way to adjust PIDs solo is with a knob, but my ideal tuning setup has been one person flying and a friend on the laptop plotting and modifying gains. Then you can coordinate the type of flying, specifically test roll at different frequencies, etc. It was because of how annoying we found it to jump between the plot and param editor in QGC that I did the mockup. The other cool feature would be if changing a param (eg FW_PR_P) was annotated, but one step at a time. The overall goal is making it as easy and fast as possible to tune each loop.

Is there anything I can do to help?

ecmnet, 2016-06-18:
@dagar I think, I'll do a kind of overlay for the relevant parameters similar to the details-overlay. On top of the overlay, there could be a drop-down with the parameter-groups and below that, the selected params. This would make the parameter-tab obsolete. (Not too bad, as the parameter-tab implementation is really ugly). What do you think?

dagar, 2016-06-18:
That sounds good.

ecmnet, 2016-06-18:
Would look like this:

[image]

Maybe Spinners would also help..

dagar, 2016-06-18:
Nice.
It looks like you have access to the param metadata. Are you aware of the decimal/increment fields?
https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/fw_att_control_params.c#L78

ecmnet, 2016-06-18:
The decimals I already have (no increment, as I currently have no spinner):
https://github.com/ecmnet/MAVGCL/blob/master/MAVGCL/src/com/comino/flight/parameter/ParameterFactMetaData.java#L106

ecmnet, 2016-06-18:
This is current status for today:

[image]

ecmnet, 2016-06-19:
@dagar Another point: In SITL changing e.g. ROLL_P from 5.50 to 5.60 immediately crashes the device with heavy peaks in actual angles:

[image]

I did not expect that. Maybe you can check this.

dagar, 2016-06-19:
That's weird. Unfortunately the param changes aren't logged. Which SITL config are you using? 
If it's gazebo iris then you could comment out the line that sets the param every startup, and see what MC_ROLL_P is actually set to. `param show MC_ROLL_P`
https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/rcS_gazebo_iris#L33

ecmnet, 2016-06-19:
It wasn't gazebo but just a quick jMAVSim test

ecmnet, 2016-06-20:
@dagar Simple LogMessage annotation in time charts include also parameter changes (you have to switch to 10sec frame resolution to see annotations; text be click):

[image]

ecmnet, 2016-06-22:
@dagar I added now actuator control target. What is the corresponding actual? Is it servo_raw output normalized by (x-1500)/1000?

dagar, 2016-06-22:
Yes it's normalized, but the PWM value is going to depend on the specific mixer. Ideally someday I'd like to calibrate it.

ecmnet, 2016-06-26:
Closing this for now. Further key figures with general key-figure refactoring.

closed 2016-06-26 by ecmnet