ecmnet/MAVGCL #59: Release build does not contain dependencies

ecmnet, 2016-06-24:
@dagar Currently the release build does not contain the dependent libraries. Could you fix this by building a zip file containing all necessary files?

dagar, 2016-06-24:
We could make that a generic deploy in the ant script, then point travis at the resulting zip.

ecmnet, 2016-06-24:
The ant-script already contains a packaging for OSX. Could we publish that instead? I could add targets for Linux and perhaps Windows as well then.

dagar, 2016-06-24:
We should probably do all of the above. Some people will want packages, some people won't.

The OSX packaging is already on travis-ci, I just haven't uploaded it yet because the file is named MAVGAnalysis-1.0.dmg. We can either try to be clever and auto name from the git tag, or rename it generically. 
https://travis-ci.org/ecmnet/MAVGCL/jobs/140031032#L575

Can you only create packages for the same system as the build host? Can we use a travis-ci linux builder to create them all?

ecmnet, 2016-06-24:
I don't think its called MAVGAnalysis-1.0.dmg, but MAVGAnalysis.dmg. So I don't understand exactly what you mean. I added the javafx build targets for Linux and Windows, but I cannot test those, as the build runs only on the os of my host (OS X).

dagar, 2016-06-24:
https://travis-ci.org/ecmnet/MAVGCL/jobs/140031032#L575

    Result DMG installer for MAVGAnalysis: 
    /Users/travis/build/ecmnet/MAVGCL/MAVGCL/dist/deploy/bundles/MAVGAnalysis-1.0.dmg

ecmnet, 2016-06-24:
Ok. I added the version attribute to the script. Output should be MAVGAnalysis-0.3.dmg

ecmnet, 2016-06-24:
Could you try a Linux/Windows build?

dagar, 2016-06-25:
I've added Linux to travis-ci - #60
We'll need to add configure appveyor for Windows builds.

ecmnet, 2016-06-25:
Thanks a lot. Maybe we could skip the windows build for now, as the tool is targeted to dev mainly, which use Linux/OSX most.

dagar, 2016-06-25:
Should we also do a "fat" jar with all the dependencies?

ecmnet, 2016-06-26:
I don't see the need of a fat jar, if we make platform specific builds. I added the repo to an appveyor account, btw.

closed 2017-01-15 by dagar