ecmnet/MAVGCL #62: Key figure refactoring

ecmnet, 2016-06-26:


ecmnet, 2016-07-05:
@dagar I am currently thinking of a redesign of the key figure definition / selection to consolidate keayfigures acquired via MAVlink and those available in the logs turning MAVGAnalysis a bit into FlightLog:
- complete mapping PX4Log key figures
- consolidate MAVLink key figures to their corresponding PX4Log key figures
- mapping via metadata
What is your opinion? Is it worth the effort for MAVGAnalysis?

dagar, 2016-07-05:
It would certainly be nice if possible, I just don't know how you could possibly keep them in sync. The bulk of the sdlog2 and mavlink modules are just mapping uORB messages to log messages and mavlink messages. Plus things aren't one to one. There's a lot of conversion.

https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L1549
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L747

Are you aware of the new logger effort? It's only in master at the moment, but it directly logs uorb messages.
https://github.com/PX4/Firmware/tree/master/src/modules/logger

ecmnet, 2016-07-05:
@dagar Perhaps you could have a look to the DataModel metadata, I've created:
https://github.com/ecmnet/MAVGCL/blob/ecm_new_datamodel/MAVGCL/src/com/comino/flight/model/AnalysisDataModelMetaData.xml. 
It allows a dynamic definition of key figures for analysis, grouping and attaching different source definitions (currently implemented for PX4Logs and the MAVComm data model. Additionally attributes like output mask, unit, description can be added. To maintain this, especially to define the corresponding PX4Log key figures, I could need some help. What do you think about it? Interested in helping here?

An initial implementation for the corresponding helper classes you can find here:
https://github.com/ecmnet/MAVGCL/tree/ecm_new_datamodel/MAVGCL/src/com/comino/flight/model

dagar, 2016-07-06:
I'm not sure I properly understand the need for key figures. I don't want to derail your efforts, but could we instead update the mavlink spec and uorb definitions to have the cleanly defined metadata you need? That way as mavlink and PX4 continue to develop and evolve MAVGCL would immediately support the new stuff.

ecmnet, 2016-07-06:
Generally I agree, although the static model currently used, has its limitations. Switching to uorb and mavlink metadata could ne a next step then.

ecmnet, 2016-07-13:
@dagar Just for your info: New logger files are now supported. I decided still to use the key figure mapping for several reasons:

- ULog key figures are not self describing and hard to understand
- Consolidation between MAVLink, PX4Log and ULog key figures
- Additional key figures from companion
- I plan to allow users loading the metadata enabling multiple files for selected scopes

dagar, 2016-07-13:
Nice! I'm actually really looking forward to the new logger.

closed 2016-07-18 by ecmnet