ecmnet/MAVGCL #66: Appveyor Github Authentication token

dagar, 2016-06-27:
@ecmnet In order for appveyor to upload to Github releases you have to generate a token on github and give it to appveyor. I don't have access to do this.

Instructions here "GitHub authentication token" - https://www.appveyor.com/docs/deployment/github

ecmnet, 2016-06-27:
@dagar: The encrypted token is Hq5VDnfPj9yDRim502Eyhn7rFSgjfo3TiRs2xM+qP/wm/Wav6q/XeVNIGEEdXbtN

dagar, 2016-06-27:
Can you tell if deploy_windows is actually working on appveyor? It's not printing the output file like the other platforms.
https://ci.appveyor.com/project/ecmnet/mavgcl/build/28#L1767

ecmnet, 2016-06-27:
No, seems to skip the install bundle step. I have no chance to get this working here on OS X. But I saw, that C:\Program Files (x86)\Inno Setup 5; should be added to the path in order to run iscc.exe

ecmnet, 2016-06-27:
Ok, now a *.msi file is created. But I cannot test it.

PR #67 (2016-06-28): appveyor deploy tags to github releases
- closes #66
commit 6887a941 (2016-06-28): appveyor deploy to github releases (#67)

- closes #66
closed 2016-06-28 by dagar