ecmnet/MAVGCL #89: Support local UDP relay via custom UDP bind port

NaterGator, 2016-08-19:
I would like to add functionality to QGroundControl which allows it transparently multiplexing traffic through to local applications. 

Currently MAVGAnalysis doesn't have the capability to configure the port on which it locally binds (it always tries to bind 14550 which QGC needs to bind to accept incoming MAV traffic). 

If we can configure the local port for MAVGAnalysis I can wire this up so the two applications can exist side-by-side with QGC managing the communication links.

ecmnet, 2016-08-19:
Try File -> Preferences to setup the port

Eike

NaterGator, 2016-08-19:
@ecmnet that only modifies the port variable [here](https://github.com/ecmnet/MAVGCL/blob/3506504275912cc2c5a2f094b2681c32488656b9/MAVGCL/src/com/comino/flight/MainApp.java#L138-L155). What I'm proposing is allowing for a randomized bound port number in MAVGAnalysis so it can act like a client to the GCS application.

NaterGator, 2016-08-19:
I'm going to work on this and submit PRs when things are ready.

closed 2016-08-19 by NaterGator

ecmnet, 2016-08-20:
Perfect 😄 I'll work on MAVGCL Monday again.

Eike