ecmnet/MAVGCL #90: Plot shows average values instead of picks

matanhavi, 2016-08-22:
Many times the user want to see the picks of some values instead of the average when the plot is zoomed out.
Add on option to show plot either on average/picks mode

ecmnet, 2016-08-22:
Could you Share an example via uploaded log?

ecmnet, 2016-08-30:
Datapoint are not averaged at all. If time resolution is lower than the acquisition rate of 50ms, a spike detector determines min/max. Refer to https://github.com/ecmnet/MAVGCL/blob/master/MAVGCL/src/com/comino/flight/widgets/charts/line/LineChartWidget.java#L681

matanhavi, 2016-09-05:
for example, on some zoom you can see about 25.4 max current.
[image]
but when zooming in you can see the 30A max:
[image]

ecmnet, 2016-09-07:
@matanhavi Was that during online acquisition or while analyzing a log file? Could you please share the log via http://logs.uaventure.com?

matanhavi, 2016-09-28:
This is during importing px4Log, 
sorry, I can't send logs

closed 2016-11-14 by ecmnet