ecmnet/MAVGCL #97: Cannot connect realtime via UDP

pziyang, 2017-03-10:
Hi, First of all, great job on this beautiful program (i'm using it to analyse logs offline). 

However, I cannot seem to get MAVGCL (v0.622) to connect realtime via Mavlink.

I am using pixhawk/radio ---> USB/serial [Laptop] --> MavProxy [Laptop] ---> UDP:14500

both QGC and MissionPlanner can connect to the UDP but not MAVGCL. I have already set the ip/port in the preference to 127.0.0.1:14500. Is there something I'm doing wrong?

Platform Win7 x64

ecmnet, 2017-03-14:
Have you tried to use port 14555?

pziyang, 2017-03-16:
The usage is going to be 

pixhawk/radio ---> USB/serial [Laptop] --> MavProxy [Laptop] ---> UDP:14550 (QGC)
                                                                              |------------------> UDP:14550 (MAVGCL)

You mean change the MAVGCL to use 14555?

ecmnet, 2017-03-18:
I am not familiar with this setup. Does it connect without QGC?

pziyang, 2017-03-31:
No, it doesn't. Even with 

pixhawk/radio ---> USB/serial [Laptop] --> MavProxy [Laptop] ---> UDP:14551 (Mavlink)

I have set the IP preference to 14551

ecmnet, 2017-04-03:
Sounds a bit strange as I have no issues to connect via UDP. I'll give it a try with MavProxy.

pziyang, 2017-04-03:
mavproxy.exe --master=com3 --baudrate=57600 --out=udp:127.0.0.1:14550 --out=udp:127.0.0.1:14551

This is the mavproxy forwarding.

ecmnet, 2017-04-03:
Thanks for your input. Currently I cannot setup mavproxy currently, as I am running OS X. On Linux as well on OS X MAVGCL supports direct USB connection. But I fear in your case this won't help as you're running windows (which I do not have).

I also fear, that 57600 baud is far too low to take advantage of logging realtime data. I personally run a companion on my vehicle, which is connected to PX4 with 921600 baud and serves UDP packets via WiFi.

ecmnet, 2017-04-03:
Try Version 0.625, which fixes the port preferences. There you can now specify the bind port (second port option in preferences)

closed 2017-04-22 by ecmnet