ecmnet/MAVGCL #104: Export to my google map

imrasp, 2017-11-02:
I am searching for the way to get gps data from px4 log (ulog) either export to google map or excel file (contain lat lon alt). Is there any option in MAVGCL or tool for this case.

Thank you in advance for any kind of help.

ecmnet, 2017-11-02:
No, not with MAVGCL, but you can use the MAVLINK related python script available from the PX4 community to start with your own implementation on this.

closed 2017-11-02 by ecmnet

mrteera, 2017-11-02:
Maybe you can try

```
pip install pyulog
```

Create a python file e.g. get_gps.py
```
from pyulog import extract_gps_dump

extract_gps_dump.main()
```

And run `python get_gps.py log.ulg`

Reference: https://github.com/PX4/pyulog