ecmnet/MAVGCL #115: running on ubuntu 18.04

kobytg17, 2019-02-27:
Hi,

I'm trying to run MAVGCL on Ubuntu 18.04. I've installed Oracle JDK with javafx, and I'm getting the following exception:

```
java -jar MAVGAnalysis.jar
Gtk-Message: 11:41:29.994: Failed to load module "canberra-gtk-module"
Exception in thread "JavaFX Application Thread" Exception in thread "main" java.lang.NoClassDefFoundError: org/mavlink/messages/MAVLinkMessage
        at java.lang.Class.forName0(Native Method)
        at java.lang.Class.forName(Class.java:348)
        at com.sun.javafx.application.LauncherImpl.lambda$launchApplicationWithArgs$413(LauncherImpl.java:352)
        at com.sun.javafx.application.PlatformImpl.lambda$runAndWait$399(PlatformImpl.java:326)
        at com.sun.javafx.application.PlatformImpl.lambda$null$397(PlatformImpl.java:295)
        at java.security.AccessController.doPrivileged(Native Method)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$398(PlatformImpl.java:294)
        at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
        at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
        at com.sun.glass.ui.gtk.GtkApplication.lambda$null$203(GtkApplication.java:139)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.ClassNotFoundException: org.mavlink.messages.MAVLinkMessage
        at java.net.URLClassLoader.findClass(URLClassLoader.java:382)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:349)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
        ... 11 more
java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:767)
Caused by: java.lang.NullPointerException
        at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:383)
        at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
        ... 5 more
```

Any help is appreciated!

ecmnet, 2019-02-27:
It seems as if macomb.jar is not in your execution class path. Did you compile MAVGCL by yourself?

kobytg17, 2019-02-27:
Thanks for the quick reply..

No, I've downloaded it as an artifact from the binary releases. I also tried running
`java -jar MAVGAnalysis.jar -cp <lib_dir>`
and ran into the same error

ecmnet, 2019-02-28:
Could you try the latest build (0.801)?

kobytg17, 2019-02-28:
Thanks again for you response.. Downloaded the new build, and I'm still getting the same error:

```
java -cp MAVGCL/MAVGCL/lib -jar MAVGAnalysis.jar 
Gtk-Message: 09:58:25.089: Failed to load module "canberra-gtk-module"
Exception in thread "JavaFX Application Thread" Exception in thread "main" java.lang.NoClassDefFoundError: org/mavlink/messages/MAVLinkMessage
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:348)
	at com.sun.javafx.application.LauncherImpl.lambda$launchApplicationWithArgs$413(LauncherImpl.java:352)
	at com.sun.javafx.application.PlatformImpl.lambda$runAndWait$399(PlatformImpl.java:326)
	at com.sun.javafx.application.PlatformImpl.lambda$null$397(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$398(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$203(GtkApplication.java:139)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.ClassNotFoundException: org.mavlink.messages.MAVLinkMessage
	at java.net.URLClassLoader.findClass(URLClassLoader.java:382)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:349)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 11 more
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:767)
Caused by: java.lang.NullPointerException
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:383)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	... 5 more
```

ecmnet, 2019-02-28:
Could you ensure that the libs that are mentioned in build.xml are included in the CP:

<copy todir="dist">
			<fileset dir="lib/">
				<include name="mavcomm.jar" />
				<include name="mavbase.jar" />
				<include name="jSerialComm-1.3.11.jar" />
				<include name="jna-4.1.0.jar" />
				<include name="JCSG.jar" />
				<include name="medusa-7.4.jar" />
				<include name="gson-2.8.5.jar" />
				<include name="jinput.jar" />
				<include name="exp4j-0.4.7.jar" />
				<include name="jfxchartext-1.1.jar" />
				<include name="core-0.29.jar" />
				<include name="dense64-0.29.jar" />
				<include name="jcodec-0.1.9.jar" />
				<include name="jcodec-javase-0.1.9.jar" />
				<include name="georegression-0.10.jar" />
				<include name="ObjImport.jar" />
			</fileset>
		</copy>

ecmnet, 2019-02-28:
```
<copy todir="dist">
			<fileset dir="lib/">
				<include name="mavcomm.jar" />
				<include name="mavbase.jar" />
				<include name="jSerialComm-1.3.11.jar" />
				<include name="jna-4.1.0.jar" />
				<include name="JCSG.jar" />
				<include name="medusa-7.4.jar" />
				<include name="gson-2.8.5.jar" />
				<include name="jinput.jar" />
				<include name="exp4j-0.4.7.jar" />
				<include name="jfxchartext-1.1.jar" />
				<include name="core-0.29.jar" />
				<include name="dense64-0.29.jar" />
				<include name="jcodec-0.1.9.jar" />
				<include name="jcodec-javase-0.1.9.jar" />
				<include name="georegression-0.10.jar" />
				<include name="ObjImport.jar" />
			</fileset>
		</copy>
```

bys1123, 2019-04-13:
I got error like this:
```
stone@ubuntu:~/Downloads$ sudo java -jar MAVGAnalysis.jar
Error: Could not find or load main class com.comino.flight.MainApp
Caused by: java.lang.NoClassDefFoundError: javafx/application/Application
```

bys1123, 2019-04-15:
[image]
Successful! Thanks for your work, super cool UI design.

WhiteBearSpirit, 2020-04-10:
I got same errors on both Windows and Linux. Tried Java 14(both) and 10(linux only) with latest javaFX.
```
D:\Programs>"C:\Program Files\Java\jdk-14\bin\java.exe" --module-path %PATH_TO_FX% --add-modules=javafx.base,javafx.controls,javafx.fxml,javafx.graphics,javafx.media,javafx.swing,javafx.web -jar MAVGAnalysis.jar
Missing JavaFX application class com.comino.flight.MainApp
```

George-avy, 2020-06-10:
I have the same problem on Ubuntu 18.04 and Java 11. I'm not sure what binary file I should download or what command I should run.

davidefolli93, 2020-08-12:
Hi @bys1123 

how did you managed to run the app on Windows?
I'm trying to run both the precomipled binaries (those available in the link from section **Binaries**: in the README file) (trying to run on Windows and Ubuntu 20.04) and those compiled by myself (trying to compile and run on ubuntu 20.04).

I am not able to run nor to compile neither on Windows or on Ubuntu.

When i try to run precompiled binaries on Windows I keep getting an Exception related to comino class
```
PS C:\Users\david\Downloads> java -jar .\MAVGAnalysis.jar
Exception in thread "JavaFX Application Thread" Exception in thread "main" java.lang.NoClassDefFoundError: com/comino/mav/control/IMAVController
        at java.lang.Class.forName0(Native Method)
```

When I try to compile on Ubuntu 20.04 I keep getting errors related to JavaFX even after installation of openjfx. Comino related errors are also present.

Thank you.
Davide

bys1123, 2020-11-19:
@davidefolli93 I think openjfx is incompatible, you should find old oracle java8 to fix this if you don't want to change the code.