ecmnet/MAVGCL #123: Is it possiblt to have binaries for Windows?

dmitry1945, 2020-08-12:
Dear Madam/Sir,

The tool looks very good, like a software should looks. 👍 
Is it possible to have a Windows 10 x64 binaries for that?
I have try to build, got the errors like this: "error: package javafx.application does not exist".
And stop to try. Sure, I can build on VM on Ubuntu, but better to have Windows binaries. :)

Thanks,
Dmitry

davidefolli93, 2020-08-13:
Hi @dmitry1945 ,

Same error also building on Ubuntu even if openjdk-8-jdk and openjfx packages are installed.

> Buildfile: /mnt/c/Users/david/sources/MAVGCL/MAVGCL/build.xml
>   [taskdef] Could not load definitions from resource com/sun/javafx/tools/ant/antlib.xml. It could not be found.
> 
> clean:
>     [mkdir] Created dir: /mnt/c/Users/david/sources/MAVGCL/MAVGCL/dist
>     [mkdir] Created dir: /mnt/c/Users/david/sources/MAVGCL/MAVGCL/dist/deploy
>     [mkdir] Created dir: /mnt/c/Users/david/sources/MAVGCL/MAVGCL/bin
> 
> metadata:
>      [echo] Current build number:0
> [propertyfile] Updating property file: /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/build.info
> 
> compile:
>      [copy] Copying 53 files to /mnt/c/Users/david/sources/MAVGCL/MAVGCL/bin
>      [copy] Copied 83 empty directories to 44 empty directories under /mnt/c/Users/david/sources/MAVGCL/MAVGCL/bin
>     [javac] Compiling 193 source files to /mnt/c/Users/david/sources/MAVGCL/MAVGCL/bin
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FXMLLoadHelper.java:36: error: package javafx.application does not exist
>     [javac] import javafx.application.Application;
>     [javac]                          ^
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FXMLLoadHelper.java:37: error: package javafx.application.Preloader does not exist
>     [javac] import javafx.application.Preloader.ProgressNotification;
>     [javac]                                    ^
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FXMLLoadHelper.java:38: error: package javafx.fxml does not exist
>     [javac] import javafx.fxml.FXMLLoader;
>     [javac]                   ^
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FXMLLoadHelper.java:39: error: package javafx.scene does not exist
>     [javac] import javafx.scene.Node;
>     [javac]                    ^
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FXMLLoadHelper.java:44: error: cannot find symbol
>     [javac]     private static Application app = null;
>     [javac]                    ^
>     [javac]   symbol:   class Application
>     [javac]   location: class FXMLLoadHelper
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FXMLLoadHelper.java:46: error: cannot find symbol
>     [javac]     public static void setApplication(Application application) {
>     [javac]                                       ^
>     [javac]   symbol:   class Application
>     [javac]   location: class FXMLLoadHelper
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FXMLLoadHelper.java:50: error: cannot find symbol
>     [javac]     public static void load(Node obj,String fxml) {
>     [javac]                             ^
>     [javac]   symbol:   class Node
>     [javac]   location: class FXMLLoadHelper
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FirstPreloader.java:35: error: package javafx.application does not exist
>     [javac] import javafx.application.Platform;
>     [javac]                          ^
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FirstPreloader.java:36: error: package javafx.application does not exist
>     [javac] import javafx.application.Preloader;
>     [javac]                          ^
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FirstPreloader.java:37: error: package javafx.scene does not exist
>     [javac] import javafx.scene.Scene;
>     [javac]                    ^
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FirstPreloader.java:38: error: package javafx.scene.control does not exist
>     [javac] import javafx.scene.control.ProgressBar;
>     [javac]                            ^
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FirstPreloader.java:39: error: package javafx.scene.image does not exist
>     [javac] import javafx.scene.image.Image;
>     [javac]                          ^
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FirstPreloader.java:40: error: package javafx.scene.image does not exist
>     [javac] import javafx.scene.image.ImageView;
>     [javac]                          ^
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FirstPreloader.java:41: error: package javafx.scene.layout does not exist
>     [javac] import javafx.scene.layout.VBox;
>     [javac]                           ^
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/FirstPreloader.java:42: error: package javafx.scene.paint does not exist
>     [javac] import javafx.scene.paint.Color;
[... 34 lines cut ...]
>     [javac]     public void handleApplicationNotification(PreloaderNotification pn) {

when you manage to solve these javafx errorz, you may come into new ones, like these:

>     [javac]   symbol:   class PreloaderNotification
>     [javac]   location: class FirstPreloader
>     [javac] /mnt/c/Users/david/sources/MAVGCL/MAVGCL/src/com/comino/flight/MainApp.java:43: error: cannot access MAV_CMD
>     [javac] import org.mavlink.messages.MAV_CMD;
>     [javac]                            ^
>     [javac]   bad class file: /mnt/c/Users/david/sources/MAVGCL/MAVGCL/lib/mavcom-0.0.1.jar(org/mavlink/messages/MAV_CMD.class)
[... cut ...]
>     [javac]     Please remove or make sure it appears in the correct subdirectory of the classpath.

The jar binaries may be cross-platform (unless that some native libraries have been used).
As you can see from this comment
[https://github.com/ecmnet/MAVGCL/issues/115#issuecomment-483189806]()
gituser (https://github.com/bys1123)
Managed to run MAVGCL on windows 

Compilation errors may be related to build.xml file as suggested by @ecmnet or to build environment setup.

If you manage to build binaries on ubuntu, please report the build process used.

Thank you.
Davide

BazookaJoe1900, 2021-04-17:
Asking here too

closed 2021-05-19 by ecmnet