ecmnet/MAVGCL #125: Build issue using JAVA 8

Stevesies, 2020-11-19:
Hello, I am trying to build from source (using release rev 0.806) with JAVA 8 JDK.  During the build I receive an error from the mavcom-0.0.1.jar file reading 
"[javac]   bad class file: /home/px/MAVGCL/MAVGCL-0.806/MAVGCL/lib/mavcom-0.0.1.jar(org/mavlink/messages/MAV_CMD.class)
  [javac]     class file has wrong version 54.0, should be 52.0
  [javac]     Please remove or make sure it appears in the correct subdirectory of the classpath.
"
May I have some assistance rebuilding the mavcom source using JAVA 8 (an alternate Java 8 compiled jar or some build instructions?). I am developing in both Ubuntu 18 and Windows 10 (same issue on both platforms).  Note I was able to build an earlier release 0.691 using this version. I am interested in using 8 as it was mentioned as being compatible and still has an LTS release (not to mention as of rev 11 and newer the JDK has changed substantially). Side question, are you planning to refactor the code to support newer versions of Java?

ecmnet, 2020-11-23:
Unfortunately, MAVGCL depends on the static compiled library mavcom.jar, which I compiled with Java 11. You would have to build mavcom.jar on your own until I migrate MAVGCL to maven.

Stevesies, 2020-11-23:
For use with ver 0.806 of MAVGCL is there a recommended version of mavcom? Is it generally backward compatible? Should I build from the current master repos? Do you have any information regarding how you generally build the jar file for mavcom?

Another question as well, are you planning on migrating this project (MAVGCL) to Java 11 and/or newer?

Thank you for you help

ecmnet, 2020-11-29:
I just released 0.820 - you could use this version with the latest mavcom.

closed 2021-05-19 by ecmnet