ecmnet/MAVGCL #126: Compilation under Linux(Ubuntu 20.04) failing

dookei, 2020-11-26:
Hi there, 
I am currently facing issues compiling under Ubuntu 
```
dookei@dookei-laptop:~/portable/MAVGCL/MAVGCL$ ant all
Buildfile: /home/dookei/portable/MAVGCL/MAVGCL/build.xml
   [script] Warning: Nashorn engine is planned to be removed from a future JDK release
  [taskdef] Could not load definitions from resource com/sun/javafx/tools/ant/antlib.xml. It could not be found.

clean:
   [delete] Deleting directory /home/dookei/portable/MAVGCL/MAVGCL/dist
    [mkdir] Created dir: /home/dookei/portable/MAVGCL/MAVGCL/dist
    [mkdir] Created dir: /home/dookei/portable/MAVGCL/MAVGCL/dist/deploy
   [delete] Deleting directory /home/dookei/portable/MAVGCL/MAVGCL/bin
    [mkdir] Created dir: /home/dookei/portable/MAVGCL/MAVGCL/bin

metadata:
     [echo] Current build number:6
[propertyfile] Updating property file: /home/dookei/portable/MAVGCL/MAVGCL/src/com/comino/flight/build.info

compile:
     [copy] Copying 55 files to /home/dookei/portable/MAVGCL/MAVGCL/bin
     [copy] Copied 86 empty directories to 45 empty directories under /home/dookei/portable/MAVGCL/MAVGCL/bin
    [javac] Compiling 201 source files to /home/dookei/portable/MAVGCL/MAVGCL/bin
    [javac] /home/dookei/portable/MAVGCL/MAVGCL/src/org/json/simple/JSONValue.java:44: warning: [dep-ann] deprecated item is not annotated with @Deprecated
    [javac] 	public static Object parse(Reader in){
    [javac] 	                     ^
    [javac] /home/dookei/portable/MAVGCL/MAVGCL/src/org/json/simple/JSONValue.java:74: warning: [dep-ann] deprecated item is not annotated with @Deprecated
    [javac] 	public static Object parse(String s){
    [javac] 	                     ^
    [javac] Note: Some input files use or override a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.
    [javac] Note: Some input files use unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.
    [javac] 2 warnings

build:

BUILD FAILED
/home/dookei/portable/MAVGCL/MAVGCL/build.xml:54: Problem: failed to create task or type javafx:com.sun.javafx.tools.ant:jar
Cause: The name is undefined.
Action: Check the spelling.
Action: Check that any custom tasks/types have been declared.
Action: Check that any <presetdef>/<macrodef> declarations have taken place.
No types or tasks have been defined in this namespace yet

```
This comes from the package JavaFX.I installed Java with Fx from **sdk install java 11.0.2-zulufx**.
By using [sdk man](https://sdkman.io/install) . I am able to compile some Java Fx examples 

```
java --version 
openjdk 11.0.9.1 2020-11-04
OpenJDK Runtime Environment (build 11.0.9.1+1-Ubuntu-0ubuntu1.20.04)
OpenJDK 64-Bit Server VM (build 11.0.9.1+1-Ubuntu-0ubuntu1.20.04, mixed mode, sharing)
```
Is there any step guide or dependencies that need to be installed ? 
I would like very much to give this a go. 
Much appreciated for any guidance.

ecmnet, 2020-11-29:
Have a look here: https://docs.oracle.com/javafx/2/deployment/javafx_ant_tasks.htm

dookei, 2020-12-07:
Thanks, which JDK version do you recommend to run it under Ubuntu 20.04?

ecarter-anduril, 2021-04-13:
I just changed the mvn path in build.xml and I'm still having trouble with the build under Ubuntu 20.04. Anyone managed to get this to work?

```
$ java -version
openjdk version "1.8.0_282"
OpenJDK Runtime Environment (Zulu 8.52.0.23-CA-linux64) (build 1.8.0_282-b08)
OpenJDK 64-Bit Server VM (Zulu 8.52.0.23-CA-linux64) (build 25.282-b08, mixed mode)
```

```
$ ant deploy_linux
Buildfile: /home/ecarter/workspace/MAVGCL/MAVGCL/build.xml

metadata:
     [echo] JDK used: /home/ecarter/.sdkman/candidates/java/8.0.282.fx-zulu/
     [echo] Current build number:5
[propertyfile] Updating property file: /home/ecarter/workspace/MAVGCL/MAVGCL/src/main/java/com/comino/flight/build.info

mvn:
     [exec] [INFO] Scanning for projects...
     [exec] [INFO] 
     [exec] [INFO] -------------------------< com.comino:MAVGCL >--------------------------
     [exec] [INFO] Building MAVGCL2 0.8.23
     [exec] [INFO] --------------------------------[ jar ]---------------------------------
     [exec] [WARNING] The POM for mavbase:mavbase:jar:0.0.1 is missing, no dependency information available
     [exec] [WARNING] The POM for jcodec:jcodec-javase:jar:0.1.9 is missing, no dependency information available
     [exec] [WARNING] The POM for com.comino:mavmap:jar:0.0.1 is missing, no dependency information available
     [exec] [WARNING] The POM for com.comino:mavcom:jar:0.0.1 is missing, no dependency information available
     [exec] [WARNING] The POM for objimport:objimport:jar:0.8.0 is missing, no dependency information available
     [exec] [INFO] ------------------------------------------------------------------------
     [exec] [INFO] BUILD FAILURE
     [exec] [INFO] ------------------------------------------------------------------------
     [exec] [INFO] Total time:  0.301 s
     [exec] [INFO] Finished at: 2021-04-13T15:47:40-07:00
     [exec] [INFO] ------------------------------------------------------------------------
     [exec] [ERROR] Failed to execute goal on project MAVGCL: Could not resolve dependencies for project com.comino:MAVGCL:jar:0.8.23: The following artifacts could not be resolved: com.comino:mavmap:jar:0.0.1, com.comino:mavcom:jar:0.0.1: Failure to find com.comino:mavmap:jar:0.0.1 in https://jitpack.io was cached in the local repository, resolution will not be reattempted until the update interval of jitpack.io has elapsed or updates are forced -> [Help 1]
     [exec] [ERROR] 
     [exec] [ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
     [exec] [ERROR] Re-run Maven using the -X switch to enable full debug logging.
     [exec] [ERROR] 
     [exec] [ERROR] For more information about the errors and possible solutions, please read the following articles:
     [exec] [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException
     [exec] Result: 1

deploy_linux:
No base JDK. Package will use system JRE.
No base JDK. Package will use system JRE.
Bundler DEB Installer skipped because of a configuration problem: Main application jar is missing.  
Advice to fix: Make sure to use fx:jar task to create main application jar.

BUILD SUCCESSFUL
Total time: 2 seconds
```

dookei, 2021-04-16:
Sorry no luck neither...

closed 2021-04-16 by dookei