ecmnet/MAVGCL #131: dmg file issue/build on macOS

tomcattigerkkk, 2021-06-15:
Hi, 

I've downloaded the dmg for my macOS. The mac reports that the MAVGAnalysis is damaged. One can only delete the app. 

I've also tried to build the app from the source. However, when I execute 

jlink --no-header-files --no-man-pages --compress=2 --strip-debug --add-modules java.se,jdk.httpserver,javafx.controls,javafx.fxml --output ./custom_jre --module-path ./../jmods

the error:

ERROR: Hash of javafx.base (6008e1e884f1ca80022c2dbcd87401b806f082ea0b281f11f70abe6a41eef23c) differs to expected hash (aff95144c7fc93a2d04d66a171697c7185d35247c6cebd8d94d0ad5cfb1f8b8c) recorded in java.base
java.lang.module.FindException: Hash of javafx.base (6008e1e884f1ca80022c2dbcd87401b806f082ea0b281f11f70abe6a41eef23c) differs to expected hash (aff95144c7fc93a2d04d66a171697c7185d35247c6cebd8d94d0ad5cfb1f8b8c) recorded in java.base
	at java.base/java.lang.module.Resolver.findFail(Resolver.java:900)
	at java.base/java.lang.module.Resolver.checkHashes(Resolver.java:478)
	at java.base/java.lang.module.Resolver.finish(Resolver.java:377)
	at java.base/java.lang.module.Configuration.<init>(Configuration.java:140)
	at java.base/java.lang.module.Configuration.resolve(Configuration.java:423)
	at java.base/java.lang.module.Configuration.resolve(Configuration.java:255)
	at jdk.jlink/jdk.tools.jlink.internal.Jlink$JlinkConfiguration.resolve(Jlink.java:217)
	at jdk.jlink/jdk.tools.jlink.internal.JlinkTask.createImageProvider(JlinkTask.java:489)
	at jdk.jlink/jdk.tools.jlink.internal.JlinkTask.createImage(JlinkTask.java:399)
	at jdk.jlink/jdk.tools.jlink.internal.JlinkTask.run(JlinkTask.java:271)
	at jdk.jlink/jdk.tools.jlink.internal.Main.run(Main.java:54)
	at jdk.jlink/jdk.tools.jlink.internal.Main.main(Main.java:33)

I use jlink 16.0.1

Thanks.

ecmnet, 2021-06-25:
You might give a more recent build a try.

tomcattigerkkk, 2021-07-06:
@ecmnet  Thanks. It works now on macOS.

closed 2021-07-06 by tomcattigerkkk