ecmnet/MAVGCL #134: Support for Linux

dookei, 2021-08-13:
Hi there @ecmnet ,
great job with this package. I am giving this another try under Linux (Ubuntu 20.04 LTS). There seems to be still alot of issues for the compilation.

## The System

OS: 20.04.1-Ubuntu

kernel:  5.11.0-25-generic

Java :  java 16.0.2 2021-07-20 /usr/lib/jvm/java-16-oracle

## The steps

1.  git clone https://github.com/ecmnet/MAVGCL.git
2.  uncommented the build.xml for deploy_linux
3.  updated property name under build.xml

```xml
<!--<property name="workspace.dir" value="/Users/ecmnet/PixHawk/java/" />  -->
    	<property name="workspace.dir" value="/home/dookei/portable/MAVGCL" />
```

5.  ant deploy_linux

## The issue

```
ant deploy_linux
Buildfile: /home/dookei/portable/MAVGCL/MAVGCL-0.833/MAVGCL/build.xml

metadata:
     [echo] JDK used: ${jdk.home}
     [echo] Current build number:2
[propertyfile] Updating property file: /home/dookei/portable/MAVGCL/MAVGCL-0.833/MAVGCL/src/main/java/com/comino/flight/build.info

com:
     [exec] sh: 1: /usr/local/Cellar/maven/3.6.3_1/bin/mvn: not found
     [exec] Result: 127

mvn:
     [exec] sh: 1: /usr/local/Cellar/maven/3.6.3_1/bin/mvn: not found
     [exec] Result: 127

deploy_linux:

BUILD FAILED
/home/dookei/portable/MAVGCL/MAVGCL-0.833/MAVGCL/build.xml:69: Problem: failed to create task or type javafx:com.sun.javafx.tools.ant:deploy
Cause: The name is undefined.
Action: Check the spelling.
Action: Check that any custom tasks/types have been declared.
Action: Check that any <presetdef>/<macrodef> declarations have taken place.
No types or tasks have been defined in this namespace yet
```
Do you have ever gave it a try to compile under Linux? Is there any dependencies that I am missing?
I believe this is a great Tool but I believe Drone developers use mainly Linux for Sw dev. If this could be ported to linux would have also a boost on users PR.
Greetings

ecmnet, 2021-08-17:
I think OSX dependencies need to be removed in MAVGCL/build.xml, e.g. https://github.com/ecmnet/MAVGCL/blob/2c5ad24bbc0ed75386c909e9490c9d9081da2d04/MAVGCL/build.xml#L20
which point to a maven installation.

dookei, 2021-08-19:
Hi @ecmnet ,
thanks for your prompt reply...I am now having some dependencies issues ...
mvn:
     [exec] [INFO] Scanning for projects...
     [exec] [INFO] 
     [exec] [INFO] -------------------------< com.comino:MAVGCL >--------------------------
     [exec] [INFO] Building MAVGCL2 0.8.23
     [exec] [INFO] --------------------------------[ jar ]---------------------------------
     [exec] [WARNING] The POM for mavbase:mavbase:jar:0.0.1 is missing, no dependency information available
     [exec] [WARNING] The POM for jcodec:jcodec-javase:jar:0.1.9 is missing, no dependency information available
     [exec] [WARNING] The POM for com.comino:mavmap:jar:0.0.1 is missing, no dependency information available
     [exec] [WARNING] The POM for com.comino:mavcom:jar:0.0.1 is missing, no dependency information available
     [exec] [WARNING] The POM for objimport:objimport:jar:0.8.0 is missing, no dependency information available
     [exec] [INFO] ------------------------------------------------------------------------
     [exec] [INFO] BUILD FAILURE
     [exec] [INFO] ------------------------------------------------------------------------
     [exec] [INFO] Total time:  0.361 s
     [exec] [INFO] Finished at: 2021-08-19T13:03:22+02:00
     [exec] [INFO] ------------------------------------------------------------------------
     [exec] [ERROR] Failed to execute goal on project MAVGCL: Could not resolve dependencies for project com.comino:MAVGCL:jar:0.8.23: The following artifacts could not be resolved: com.comino:mavmap:jar:0.0.1, com.comino:mavcom:jar:0.0.1: Failure to find com.comino:mavmap:jar:0.0.1 in https://jitpack.io was cached in the local repository, resolution will not be reattempted until the update interval of jitpack.io has elapsed or updates are forced -> [Help 1]
     [exec] [ERROR] 
     [exec] [ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
     [exec] [ERROR] Re-run Maven using the -X switch to enable full debug logging.
     [exec] [ERROR] 
     [exec] [ERROR] For more information about the errors and possible solutions, please read the following articles:
     [exec] [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException
     [exec] Result: 1

Any thoughts ?

donfung, 2021-11-12:
I'm getting the same error on my Ubuntu 18.04. Did you ever find a solution?

dookei, 2021-11-12:
No... never had time to check it. 
There are quite a few dependencies and issues with this package under linux.

dookei, 2021-11-30:
Hi @ecmnet any news on the linux support?

PSandro, 2021-12-01:
Hi, I managed to build an AppImage on arch with the following steps:
```bash
git clone git@github.com:ecmnet/MAVGCL.git
git clone git@github.com:ecmnet/MAVBase.git
git clone git@github.com:ecmnet/mavcom.git
git clone git@github.com:ecmnet/mavutils.git
git clone git@github.com:ecmnet/mavmap.git
```
Go in each project directory and install to local maven repo (with `mvn clean install`) in this order:

1. `mavutils`
2. `mavcom`
3. `mavmap`
4. `MAVBase`

then run `mvn clean install` in project dir of `MAVGCL`. This will generate files in `./target/deploy`

I have Java version `openjdk 17.0.1` with `openjfx` installed, so I don't need to use jmods in `./jmods` directory. 
Therefor I modified the `package.sh` script to following:
```bash
#! /bin/bash

jlink --no-header-files --no-man-pages --compress=2 --strip-debug --add-modules java.se,jdk.httpserver,javafx.controls,javafx.fxml --output ./target/custom_jre

jpackage --dest target/application --input target/deploy/ --name MAVGAnalysis --main-jar MAVGCL-0.8.0.jar --main-class com.comino.flight.MainApp --type app-image --icon target/deploy/MAVGAnalysis.icns --runtime-image ./target/custom_jre --java-options '-Djava.library.path=./native -Djavafx.animation.pulse=60 --add-opens javafx.base/com.sun.javafx.collections=ALL-UNNAMED -XX:+UnlockExperimentalVMOptions -XX:+UseZGC -Xmx8G -XX:ZCollectionInterval=5 '
```
Notice that `--type dmg` was changed to `--type app-image`. You could also use `--type deb`. 

Run this script, you'll find the AppImage in `target/application/MAVGAnalysis/bin/MAVGAnalysis`

dookei, 2021-12-02:
Hi @PSandro thanks for your feedback and time. 
Unfortunately I was not able to go any further than the first step 
```
dookei@dookei-laptop:~/local_builds/mavgcl_local/mavutils$ mvn clean install
[ERROR] Error executing Maven.
[ERROR] java.lang.IllegalStateException: Unable to load cache item
[ERROR] Caused by: Unable to load cache item
[ERROR] Caused by: Could not initialize class com.google.inject.internal.cglib.core.$MethodWrapper
```

I also installed openjdk-17
```
java -version
openjdk version "17" 2021-09-14
OpenJDK Runtime Environment (build 17+35-Ubuntu-120.04)
OpenJDK 64-Bit Server VM (build 17+35-Ubuntu-120.04, mixed mode, sharing)
```

I set maven to use also openjdk-17
```
mvn -v
Apache Maven 3.6.3
Maven home: /usr/share/maven
Java version: 17, vendor: Private Build, runtime: /usr/lib/jvm/java-17-openjdk-amd64
Default locale: en_US, platform encoding: UTF-8
OS name: "linux", version: "5.11.0-37-generic", arch: "amd64", family: "unix"
```

I changed to maven 3.8.4
I then was able to run mvn clean install under mavutil but no mavcom
I get than this errror 

```
[INFO] 
[INFO] --- maven-compiler-plugin:3.1:testCompile (default-testCompile) @ mavcom ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 1 source file to /home/dookei/local_builds/mavgcl_local/mavcom/mavcom/target/test-classes
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  12.894 s
[INFO] Finished at: 2021-12-02T08:50:16+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.1:testCompile (default-testCompile) on project mavcom: Fatal error compiling: java.lang.IllegalAccessError: class lombok.javac.apt.LombokProcessor (in unnamed module @0x7e907c56) cannot access class com.sun.tools.javac.processing.JavacProcessingEnvironment (in module jdk.compiler) because module jdk.compiler does not export com.sun.tools.javac.processing to unnamed module @0x7e907c56 -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
```

Is there any package or dependency that I am missing ?

ecmnet, 2022-01-17:
mavcom needs to be build separately

hdbalyozsht, 2022-01-27:
> Hi, I managed to build an AppImage on arch with the following steps:
> 
> ```shell
> git clone git@github.com:ecmnet/MAVGCL.git
> git clone git@github.com:ecmnet/MAVBase.git
> git clone git@github.com:ecmnet/mavcom.git
> git clone git@github.com:ecmnet/mavutils.git
> git clone git@github.com:ecmnet/mavmap.git
> ```
> 
> Go in each project directory and install to local maven repo (with `mvn clean install`) in this order:
> 
> 1. `mavutils`
> 2. `mavcom`
> 3. `mavmap`
> 4. `MAVBase`
> 
> Change `pom.xml` of `MAVGCL` from this:
> 
> ```
> 		<dependency>
> 			<groupId>mavbase</groupId>
> 			<artifactId>mavbase</artifactId>
> 			<version>0.0.1</version>
> 		</dependency>
> ```
> 
> to this
> 
> ```
> 		<dependency>
> 			<groupId>MAVBase</groupId>
> 			<artifactId>MAVBase</artifactId>
> 			<version>0.0.1</version>
> 		</dependency>
> ```
> 
> then run `mvn clean install` in project dir of `MAVGCL`. This will generate files in `./target/deploy`
> 
> I have Java version `openjdk 17.0.1` with `openjfx` installed, so I don't need to use jmods in `./jmods` directory. Therefor I modified the `package.sh` script to following:
> 
> ```shell
> #! /bin/bash
> 
> jlink --no-header-files --no-man-pages --compress=2 --strip-debug --add-modules java.se,jdk.httpserver,javafx.controls,javafx.fxml --output ./target/custom_jre
> 
> jpackage --dest target/application --input target/deploy/ --name MAVGAnalysis --main-jar MAVGCL-0.8.0.jar --main-class com.comino.flight.MainApp --type app-image --icon target/deploy/MAVGAnalysis.icns --runtime-image ./target/custom_jre --java-options '-Djava.library.path=./native -Djavafx.animation.pulse=60 --add-opens javafx.base/com.sun.javafx.collections=ALL-UNNAMED -XX:+UnlockExperimentalVMOptions -XX:+UseZGC -Xmx8G -XX:ZCollectionInterval=5 '
> ```
> 
> Notice that `--type dmg` was changed to `--type app-image`. You could also use `--type deb`.
> 
> Run this script, you'll find the AppImage in `target/application/MAVGAnalysis/bin/MAVGAnalysis`

I did all steps. However, couldn't run it. I get this error: 

`There is no application installed for "shared library" files.`

`Allow executing file as program` is **checked**

File information: `MAVGAnalysis: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=066c15406da022e47ac2d1254dc124d88548d40a, for GNU/Linux 3.2.0, with debug_info, not stripped`

Simone-DAngelo, 2022-04-28:
> > Hi, I managed to build an AppImage on arch with the following steps:
> > ```shell
> > git clone git@github.com:ecmnet/MAVGCL.git
> > git clone git@github.com:ecmnet/MAVBase.git
> > git clone git@github.com:ecmnet/mavcom.git
> > git clone git@github.com:ecmnet/mavutils.git
> > git clone git@github.com:ecmnet/mavmap.git
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Go in each project directory and install to local maven repo (with `mvn clean install`) in this order:
> > 
> > 1. `mavutils`
> > 2. `mavcom`
> > 3. `mavmap`
> > 4. `MAVBase`
> > 
> > Change `pom.xml` of `MAVGCL` from this:
> > ```
> > 		<dependency>
> > 			<groupId>mavbase</groupId>
> > 			<artifactId>mavbase</artifactId>
> > 			<version>0.0.1</version>
> > 		</dependency>
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > to this
> > ```
> > 		<dependency>
> > 			<groupId>MAVBase</groupId>
> > 			<artifactId>MAVBase</artifactId>
> > 			<version>0.0.1</version>
> > 		</dependency>
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > then run `mvn clean install` in project dir of `MAVGCL`. This will generate files in `./target/deploy`
> > I have Java version `openjdk 17.0.1` with `openjfx` installed, so I don't need to use jmods in `./jmods` directory. Therefor I modified the `package.sh` script to following:
> > ```shell
> > #! /bin/bash
> > 
> > jlink --no-header-files --no-man-pages --compress=2 --strip-debug --add-modules java.se,jdk.httpserver,javafx.controls,javafx.fxml --output ./target/custom_jre
> > 
> > jpackage --dest target/application --input target/deploy/ --name MAVGAnalysis --main-jar MAVGCL-0.8.0.jar --main-class com.comino.flight.MainApp --type app-image --icon target/deploy/MAVGAnalysis.icns --runtime-image ./target/custom_jre --java-options '-Djava.library.path=./native -Djavafx.animation.pulse=60 --add-opens javafx.base/com.sun.javafx.collections=ALL-UNNAMED -XX:+UnlockExperimentalVMOptions -XX:+UseZGC -Xmx8G -XX:ZCollectionInterval=5 '
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Notice that `--type dmg` was changed to `--type app-image`. You could also use `--type deb`.
> > Run this script, you'll find the AppImage in `target/application/MAVGAnalysis/bin/MAVGAnalysis`
> 
> I did all steps. However, couldn't run it. I get this error:
> 
> `There is no application installed for "shared library" files.`
> 
> `Allow executing file as program` is **checked**
> 
> File information: `MAVGAnalysis: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=066c15406da022e47ac2d1254dc124d88548d40a, for GNU/Linux 3.2.0, with debug_info, not stripped`
> 
> ```
> $ ./MAVGAnalysis 
> Graphics Device initialization failed for :  es2, sw
> Error initializing QuantumRenderer: no suitable pipeline found
> java.lang.RuntimeException: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
> 	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumRenderer.getInstance(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumToolkit.init(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.tk.Toolkit.getToolkit(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.application.LauncherImpl.startToolkit(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication(Unknown Source)
> 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
> 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
> 	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
> 	at java.base/sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)
> Caused by: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
> 	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(Unknown Source)
> 	at java.base/java.lang.Thread.run(Unknown Source)
> Exception in thread "main" java.lang.reflect.InvocationTargetException
> 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
> 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
> 	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
> 	at java.base/sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)
> Caused by: java.lang.RuntimeException: No toolkit found
> 	at javafx.graphics/com.sun.javafx.tk.Toolkit.getToolkit(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.application.LauncherImpl.startToolkit(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(Unknown Source)
> 	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication(Unknown Source)
> 	... 5 more
> pure virtual method called
> terminate called without an active exception
> Aborted (core dumped)
> ```

I have the same error, did you solve it?

sfaunl, 2023-07-27:
Here is a PKGBUILD file that I've created that may help someone. Feel free to complete it and share on AUR.

```
pkgname="MAVGCL"
pkgver="0.865"
pkgrel="1"
pkgdesc="In-Flight/PX4Log/ULog Analyzer for PX4"
url="https://github.com/ecmnet/$pkgname"
arch=("x86_64")
depends=('java-runtime' 'java-environment>11')
makedepends=('maven' 'archlinux-java-run>=7'  'java-environment>11' 'java-openjfx>11')

mavmap_url="https://github.com/ecmnet/mavmap"
mavmap_hash="be2f19bf56950ce56cc40906069cc0900812cb20"
mavutils_url="https://github.com/ecmnet/mavutils"
mavutils_hash="8c995e0754cc8b48639b7e8aba86f47a3a92f698"
mavcom_url="https://github.com/ecmnet/mavcom"
mavcom_hash="2da45e86ff7dd58073009944d3817fcd097dae90"
mavbase_url="https://github.com/ecmnet/mavbase"
mavbase_hash="a85e1a65682a3fe86090f3ac52b4ebc621e41964"

source=("MAVGCL.tar.gz::$url/archive/refs/tags/V0.865.tar.gz"
        "MAVMap.tar.gz::$mavmap_url/archive/$mavmap_hash.tar.gz"
        "MAVUtils.tar.gz::$mavutils_url/archive/$mavutils_hash.tar.gz"
        "MAVCom.tar.gz::$mavcom_url/archive/$mavcom_hash.tar.gz"
        "MAVBase.tar.gz::$mavbase_url/archive/$mavbase_hash.tar.gz")

#generate with 'makepkg -g'
sha512sums=('f1a0b2c618b172771f4e9f10da7cb793a75060a8d208e671a58b87b9006e7f722c9092663477b0d043ed0736c4893c916ee4ea982f6ccd24865c7b43123b42f5'
            'f5d69cd09f971ec530e0fb54419cebef56f8b6a64ef5f399bc539433a17337aed7cf707855be074cda2da9c8ae955796561e1ea20e975ae0b5fafe34356700d5'
            '827e7b82b93d590c0738e65b706db02a9ea81179b3ec29d53ac34d92130473bac40c2ed06d508b284c94b08c82ba32c1ed0feccf3d83225a7147b2a85c19455c'
            'd75d7127f6d15a8bb0859aecfad26023b93939f239c45d7df5e44a15ee7c209562768965e0d71f042231e88ef0dad9035a814572a39c05a6efab5451706b1238'
            '195a981a35001f853b75a00a38167c3f011277aecbc325773c956980293a1275279f34f22da4b7019bb5a7d207d2c1e255c0ee05fb87a8c8ecf4376467d4a4f7')

#javahome="/usr/lib/jvm/default"
#export JAVA_HOME=${javahome}

prepare() {
  java_v="$(archlinux-java get | grep -Po '\d+')"
  if [ "$java_v" -le "11" ]; then
    echo "ERROR: Java environment needs to be set to anything above version 11. Current version: $java_v"
    exit 1
  fi
}

build() {
  echo "Building MAVUtils..."
  cd "$srcdir/mavutils-$mavutils_hash/mavutils"
  mvn clean install

  echo "Building MAVCom..."
  cd "$srcdir/mavcom-$mavcom_hash/mavcom"
  mvn clean install

  echo "Building MAVMap..."
  cd "$srcdir/mavmap-$mavmap_hash/mavmap"
  mvn clean install

  echo "Building MAVBase..."
  cd "$srcdir/MAVBase-$mavbase_hash/MAVBase"
  mvn clean install

  echo "Preparing MAVGCL..."
  cd "$srcdir/$pkgname-$pkgver/MAVGCL"

#change workspace directory to $srcdir
  #escaped_srcdir=$(echo "$srcdir" | sed 's/[\/&]/\\&/g')
  #sed -i "s/\(<property name=\"workspace\.dir\" value=\"\)[^\"]*/\1$escaped_srcdir/" build.xml
#set JAVA_HOME environment variable to correct value
  #escaped_javahome=$(echo "$javahome" | sed 's/[\/&]/\\&/g')
  #sed -i "s/\(<env key=\"JAVA_HOME\" path=\"\)[^\"]*/\1$escaped_javahome/" build.xml
#fix maven path
  #sed -i "s|/usr/local/Cellar/maven/3.9.0/bin/mvn|mvn|" build.xml

#rename mavbase to MAVBase in pom.xml
  sed -i 's/mavbase/MAVBase/g' pom.xml

  echo "Building MAVGCL..."
  cd "$srcdir/$pkgname-$pkgver/MAVGCL"
  mvn clean install

  jlink --no-header-files --no-man-pages --compress=2 --strip-debug --add-modules java.se,jdk.httpserver,javafx.controls,javafx.fxml --output ./target/custom_jre

  jpackage --dest target/application --input target/deploy/ --name MAVGAnalysis --main-jar MAVGCL-0.8.0.jar --main-class com.comino.flight.MainApp --type app-image --icon target/deploy/MAVGAnalysis.icns --runtime-image ./target/custom_jre --java-options '-Djava.library.path=./native -Djavafx.animation.pulse=60 --add-opens javafx.base/com.sun.javafx.collections=ALL-UNNAMED --add-opens javafx.base/com.sun.javafx.runtime=ALL-UNNAMED -XX:+UnlockExperimentalVMOptions -XX:+UseZGC -Xmx8G -XX:ZCollectionInterval=5 '

}

package() {
}
```