ecmnet/MAVGCL #141: How to install on Ubuntu20.04

Lov3notform3, 2022-12-16:
`openjdk version "11.0.16" 2022-07-19`
`OpenJDK Runtime Environment (build 11.0.16+8-post-Ubuntu-0ubuntu120.04)`
`OpenJDK 64-Bit Server VM (build 11.0.16+8-post-Ubuntu-0ubuntu120.04, mixed mode, sharing)`
I git clone from here. 
And uncommented the build.xml for deploy_linux. 
Then used "ant deploy_linux" .

Terminal reply `Target "mvn" does not exist in the project "MAVGAnalysis". It is used from target "deploy_linux".`

Who can help me to solve the problem.
Thanks.

avcuenes, 2022-12-22:
I live this issue but I didnt solve it

palahmd, 2023-02-11:
What worked for me (I am using MacOS):

1.  You have to download and build the following repositories from ecmnet:
        - Mavbase
        - Mavutils
        - Mavmap
        - Mavcom

2. before building the repositories, you have to modify the respective build.xml files of Mavbase, Mavutils, Mavmap, Mavcom and MAVGCL and adjust the:
       - workspace.dir
       - deploy.dir
       - JAVA_HOME path
       - path to maven (mvn). If you don't have mvn, just download it directly from apache and place it to where it makes sense

3. Build the targets one after the other with ant, with MAVGCL being the last target. Make sure you are setting the right build argument (e.g. ant build_mavbase -> see target name in the build.xml file)

andres040193ec, 2023-05-24:
Hey,
I'm using Ubuntu 20.04, what other repositories do I need, besides MAVGCL, in order to get the MAVGCL analysis working for gazebo simulation? 

Any help is appreciated.