ecmnet/MAVGCL #142: M1 Mac build fix: x64 instead of aarch64 jmods

palahmd, 2023-02-11:
Hi, 

after trying to compile MAVGCL several times for my Macbook with M1 Pro, I repeatedly got an `Error initializing QuantumRenderer: no suitable pipeline found` error message as stated in the forum post here: https://stackoverflow.com/q/53467499

I could resolve the issue by replacing the aarch64 jmods in the arm64 folder with the x64 jmods. I got the hint from the following post on stackoverflow: https://stackoverflow.com/a/70765058 

Therefore, I would suggest to replace the jmods for arm64 if other users reported to have the same problem.

ecmnet, 2023-02-12:
I run MAVGCL on M1 Pro using aarch64. Might be that you are using rosetta2 for java.

palahmd, 2023-02-12:
Might be the case. I have most of my workflow with rosetta2 because of PX4. I didn't consider this possibility. 

Do you know if there is any performance difference using rosetta2?

palahmd, 2023-02-12:
I tried to build MAVGCL without using rosetta2 for java. However, I am getting the same issues. Did you install openjdk with brew?

ecmnet, 2023-02-17:
No I directly installed the JDK. I think the key lies in the link command of the package scripts.

palahmd, 2023-03-21:
I would then suggest to mark this issue as closed